asterics/AsTeRICS #83: Update documentation for Release 2.6

deinhofer, 2015-03-23:
The following functionality/plugins are new and must be checked for a help file or a respective entry in the Developer or User Manual. Furthermore maybe some plugins should not be added to the component collection. 

update LipMouse
update CellBoard (CellBoardGUIEditor)
update EyeX
Fabi
FabiCronusMax
EasyHomeControl
EmulateFaultyPlugin (mad):
XFaceTracker (mad)
upate Arduino: uniqueId
WriteCSV
ReadCSV
StringSplitter
KinectJ4K
update StringExpander plugin
eShoe
HRVAnalysis
ECMAScriptInterpreter
LineWriter
Update Slider, MicGPI, EditBox, CellBoard, ButtonGrid, EventRouter, EventFlipFlop, MediaPlayer, Keyboard, TextSender, StringDispatcher
LineReader
StringDelay
Delay
RemoteWindow
AREWindow

Flag displayGui bei allen GUI-Komponenten

DevManual:
JavaCV (mad)
JNativeHook (mad)
SingleThreaded (mad)
REST-Functionality (Correction of current doc)
Linux Startup (mad)

ChrisVeigl, 2015-03-25:
current state is:

LipMouse  OK
CellBoard (incl. CellBoardGUIEditor)  OK
EyeX  OK
Fabi OK
FabiCronusMax OK
EasyHomeControl OK
EmulateFaultyPlugin (mad):
XFaceTracker (mad)
update Arduino: uniqueId OK
WriteCSV OK  (TBD: improve write options)
ReadCSV OK (TBD: improve read options: line-by-line, data-rate)
StringSplitter OK
KinectJ4K OK
StringExpander plugin OK
eShoe OK
HRVAnalysis OK
ECMAScriptInterpreter OK
LineWriter OK
Update Slider OK
MicGPI  OK
EditBox OK
CellBoard OK
ButtonGrid OK
EventRouter OK
EventFlipFlop OK
MediaPlayer OK
Keyboard OK
TextSender OK 
StringDispatcher OK
LineReader OK
StringDelay OK
Delay OK
RemoteWindow OK
AREWindow OK

(the TBD's can be done later, not necessarily for release 2.6 !)

kingcole, 2015-03-27:
ACSHelp.chm updated with all components mentioned in previous comment

the file "CellboardEditor.jpg" seems to be missing though

closed 2016-07-07 by deinhofer