asterics/AsTeRICS #87: Restrict access to ARE REST and/thrift ASAPI interfaces to localhost or defined IP address

deinhofer, 2015-10-19:
The access to the remote interfaces should be restrictable to localhost or a defined IP. This restriction should be the default setting of the ARE.

klues, 2017-10-20:
maybe add a property to areProperties like:
`webserver.enable.external.access=true/false`
and internally set the address to "0.0.0.0" or "localhost"

deinhofer, 2017-10-20:
On the other hand, if you can set the addresses directly to could configure it more fine grained like only for a certain interface.
But you are right, primarily we need a feature for switching off remote access for security reasons.

Let's use a property name which is consitent to the ones used for the port configuration:
ARE.webservice.port.websocket=8082
ARE.webservice.port.REST=8081

e.g.
ARE.webservice.enable.external.access=true/false

klues, 2017-10-20:
yes, a standard name is better of course :)

klues, 2018-10-16:
Working on the AsTeRICS Ergo Grid the question raised up for me again: Can I access ARE from an external webpage on the web?
The answer is yes because of these lines: https://github.com/asterics/AsTeRICS/blob/56f59f844f2a1c038ff4807ee208127d2033298b/ARE/services/WebService/src/main/java/eu/asterics/mw/webservice/ResponseFilter.java#L78

On the one hand this is very handy for our external applications that use the REST API (Demo-Pages, AsTeRICS Ergo Grid), on the other side it is a big security hole. Any page could include `areCommunicator.js` and do whatever they want with ARE, if the user opens this page.

I would recommend this solution:
Instead of disabling external access completely, we should define origins that are allowed to access the ARE. These could be the Demo pages, AsTeRICS Ergo Grid, localhost...

How to implement this: instead of always adding
`headers.add("Access-Control-Allow-Origin", "*");`
this should be done only if the request origin comes from one of the accepted origins. Unfortunately it is not possible to set multiple allowed origins directly in the `Access-Control-Allow-Origin` header. Compare the following answer on Stackoverflow: https://stackoverflow.com/questions/1653308/access-control-allow-origin-multiple-origin-domains

Maybe there should be some domains allowed by default (localhost, demos, grid) and others can be added by parameter like:
`ARE.webservice.additional.access.domain=https://mydomain.xy`

deinhofer, 2018-10-16:
This sounds very reasonable to me. 
So per default I would vote for localhost and http://asterics.github.io/.

The file areProperties should allow to add additional domains per config parameter, as you said.

Could you provide a PR?

klues, 2018-11-21:
opened #271 that resolves this issue.

allowed origins for REST API can now be specified with the ARE property:
`ARE.REST.allowed.origins`

Default allowed origins are `localhost`, `127.0.0.1`, `asterics.github.io` and `asterics-foundation.org`

klues, 2018-11-21:
Tested the implementation:
1) AsTeRICS Ergo Grid from localhost and https://asterics.github.io/AsTeRICS-Ergo-Grid/package/static/#main -> works
2) REST API Tests from https://asterics.github.io/AsTeRICS/ -> works
3) navigating to a random foreign domain which includes jquery: https://www.webpagetest.org/
and running the command `$.get('http://127.0.0.1:8081/rest/runtime/model/name', function(response) {console.log(response)})` --> does not work with new implementation, worked before

klues, 2018-11-21:
found out that my first solution does not really improve security because CORS only ensures that the Browser cannot use results of requests, e.g. the request sent by `$.get('http://127.0.0.1:8081/rest/runtime/model/name', function(response) {console.log(response)})` on https://www.webpagetest.org/

However the request is still executed on ARE, calling the corresponding method in `RestServer.java`. So any page could still trigger ARE actions, but could not use the results.

Changed the implementation and added a RequestFilter that returns HTTP 401 Unauthorized, if the origin is not allowed, see #271.
If the origin header is not set the request is allowed in order to prevent breaking other calls to the REST API that do not come from a browser.

klues, 2018-11-26:
closed by #271

closed 2018-11-26 by klues