asterics/AsTeRICS #98: AsapiSupport consolidation and use of new class ResourceRegistry

deinhofer, 2015-11-24:
Talking about the branch Licensing_Packaging_Asterics_2_8_Prep, It seems that in AsapiSupport class (https://github.com/asterics/AsTeRICS/blob/Licensing_Packaging_Asterics_2_8_Prep/ARE/middleware/src/main/java/eu/asterics/mw/are/AsapiSupport.java) we have several methods doying similar things. Let's try to consolidate them now and use new methods of ResourceRegistry, BundleManager and ComponentRepository.

# Defintion of classes:

**ResourceRegistry**: Registry to find static resources: Returns URIs of jars, images, model files, data files, ... depending on type of resource and underlying file/directory structure. URIs can be absolute or relative to the ARE.jar folder

**BundleManager**: Handles installation of OSGi bundles, registers ComponentTypes in ComponentRepository and manages several mappings: componentTypeId <-> Bundle object, componentTypeId <-> componentJarName

**ComponentRepository**: Only maps componentTypeId <-> ComponentType object without knowlege about underlying bundle.

## Note about terminology
There is a distinction between available plugins (all components supported for a platform) and installed plugins (only currently installed ones).

# What about the following AsapiSupport methods:
**public String [] getAvailableComponentTypes()**: never used? but part of Asapi interface definition: Available or currently installed component types?? --> Same as getBundleDescriptors(), use DeploymentManager.instance.getBundleManager().getInstallableBundleList();

new methods for REST interface:
**private Set<String> getInstalldedComponentJarNames**: Set of component Jar names of installed plugins --> Ask ComponentRepository about installed components and ask BundleManager about associated jar-file. Better: Add method to BundleManager that combines both steps.

**public IComponentType [] getInstalledComponents()**: Return list of installed ComponentType instances: fine

**public String getInstalledComponentsDescriptor()**: Return one xml string with bundle descriptors of all installed bundles: Ask ComponentRepository for installed components, ask
BundleManager for associated jar name, ask ResourceRegistry for bundle_descriptor URI. Let's create a method in BundleManager that returns the bundle_descriptor for a componetTypeId

**public String getCreatedComponentsDescriptors()**: What means created here? Seems to be equivalent to public List<String> getBundelDescriptors() except that it is one xml string and not a List

**public List<String> getBundelDescriptors()**: Returns a List of bundle descriptor strings (one element per bundle) of all available bundles: used by ACS Download Component Collection

**public String[] listAllStoredModels()**: Use ResourceRegistry.getInstance().getModelList()
All other model operating methods like getModel, deployModel,storeModel,...should also use ResourceRegistry.getInstance().getResource(...) or ResourceRegistry.getInstance().getModelList()

**getRESTFunctions()**: broken? only returns JSON string: {"path":"/runtime/model","consumes":"","produces":"text/xml","description":"Retrieves the currently deployed model in XML","httpRequestType":"GET","bodyParameter":""}

mariokom, 2015-12-03:
- Installed/created distinction removed
- getRESTFunctions: actually it returns all the available rest functions. The javascript API tester presents only one of them for demonstation purposes. Javascript API tester is changed to avoid confussion.

deinhofer, 2016-05-11:
Should be fixed, right? Please close it!

closed 2016-05-13 by mariokom