asterics/AsTeRICS #113: ARE in inconsistent state after upload of faulty model fails

kingcole, 2016-03-17:
When a faulty model is uploaded to the ARE (e.g. when some value that should be int reads "undefined"), the ARE remains with this non-deployable model. When someone then tries to download that model (via REST or ASAPI) the ARE throughs an error (the ACS even terminates the connection to ARE).

When there is an error during model upload, the ARE should roll back to the previous state.

Note:
To reproduce the problem, you can e.g. do this:
Upload a faulty model using e.g. the jsClientTester, then try to download that model using whatever means you like (ACS, WebACS, jsClientTester, ...).

deinhofer, 2016-03-17:
What exactly do you mean with "int reads undefined"??
Could you upload a faulty model file here? (Just drag and drop it to the issue)

kingcole, 2016-03-17:
See width and height of AREGUIWindow in attached example for explanation what I mean by undefined (this is just one of many examples of how to ruin a model of course *g*).
[faulty_model.zip](https://github.com/asterics/AsTeRICS/files/177854/faulty_model.zip)

deinhofer, 2016-05-11:
Just tested on master branch:
When trying to download a model after the deployment of the faulty model above failed the error message is:
"No model available for download"

Also if you deploy a correct model afterwards, there is no residual corrupted model.
--> seems to be fixed

closed 2016-05-11 by deinhofer