asterics/AsTeRICS #116: Windows 10 & ARE: high CPU workload with CIM

gernus, 2016-04-14:
If a CIM module (e.g. IMA GPI) is connected to a Windows 10 PC and the ARE is started the CPU workload is about 15%. If there is no CIM module connected, the CPU workload of the ARE is 0% (Demomenu). Seems to be an ARE bug.

deinhofer, 2016-04-25:
If have reproduced the problem on Win10 1511, in a VirtualBox with the Accelerometer CIM module.

According to Java Mission Control, the high CPU load is in the native call
gnu.io.RXTXPort.readByte

[image]

If you enable the receive timeout for the read operation the CPU load goes down.

```java
RXTXPort port=...
port.enableReceiveTimeout(500);
```

The question is why this is necessary for Win10 and not for the other platforms.
@cristobaldobranco: Do you see any bad side effects if we use enableReceiveTimeout? 
As port.notiveOnDataAvailable is turned on, the subsequent read call should not block but return the data.

cristobaldobranco, 2016-04-25:
I don't think that there would be a lot of side effects. As incoming data is collected in dataSource queue an occasional waking of the RXTX receive thread should not have any effect on the CIM communication service. Go ahead with the fix ;)

commit e894ab47 (2016-05-10): Fixes #116: When setting the enableReceiveTimeout, the cpu loads seems to be normal

ad gernus: Could you please test and verify that?
closed 2016-05-10 by deinhofer