asterics/AsTeRICS #131: eventChannel id inconsistency

mariokom, 2016-11-04:
Normally an eventChannel id has this format: {sourcePortID}_{targetPortID}. 
This is format is followed by all AsTeRICS models and by IEventEdge#getComponentInstanceID() method.

For some reason, in the DeploymentManager class, line 409 exists this line:

> eventTriggererPort.setEventChannelID(sourceEventPortID+" > "+eventListenerDetails.portID);

which creates and sets the eventChannelId of the upcoming event to a different format.
This inconsistency causes problems when an external event listener wants to identify a specific event. It is expecting the default format, but a different format is received.

Does anybody knows if there is a reason behind this? Should we change it?

mariokom, 2016-11-04:
In the DeploymentManager class, the correct eventChannelID is retrieved in line 364, but is not used.

deinhofer, 2016-11-09:
Do you mean because of the '>' character as seperator?
Hmm, no idea. Seems to be really strange. 
I also don't understand the meaning of IEventEdge?
Isn't IEventChannel already the channel from an eventTriggerer to an eventListener?

What happens if you correct it?
Maybe test it with a dedicated model containing all possible cardinalities of event channels (1 to 1, 1 to many, many to many)

before and after the change.

Would be good to include that test to 
https://github.com/asterics/AsTeRICS/tree/master/Tests/integrationTests/04_Model%20execution
then.

mariokom, 2016-11-10:
Correct, i am talking about the '>' character.
Ok i will test it and act according to the test results.

The IEventEdge is mostly holding information (component ids, eventPort ids) and has not functional role at runtime (it is mainly for xml creation)

mariokom, 2016-11-24:
The change of the '>' character to '_' did not produce any problems. The modification was committed.

closed 2016-11-24 by mariokom