asterics/AsTeRICS #132: REST API: new method setRuntimeComponentProperties(..., properties[]): errorMsg[]

deinhofer, 2016-11-28:
Add a new method to the REST API

```javascript
setRuntimeComponentProperties(..., properties[]): errorMsg[]
```
which let's you set several properties at once. This would be very handy, because we would like to parametrize template models which means we have to set several properties of a deployed model before starting the model.

The Web-UI would then call:
```javascript
deployFile(...);
setRuntimeComponentProperties(...);
startModel(...);
```

deinhofer, 2016-11-29:
Here you can see how it looks like, before the API enhancement:
https://github.com/asterics/AsTeRICS-Ergo/blob/master/custom/bin/ARE/data/webservice/index.html

mariokom, 2017-01-10:
Now we can change several properties of a component at once

deinhofer, 2017-01-23:
Thanks!!
I still could not test it, but will need it soon.

closed 2017-01-23 by deinhofer