asterics/AsTeRICS #133: REST API: new method storeDeployedFile(..., targetFilename)

deinhofer, 2016-11-28:
Add a new method to the REST API

```javascript
storeDeployedFile(..., targetFilename)
```
Stores the currently deployed model to the file targetFilename on the ARE. This would allow to easily save a parametrized model with the preferred settings of a user to a certain name.

The Web-UI would then call:

```javascript
deployFile(...);
setRuntimeComponentProperties(...);
startModel(...);
storeDeployedModel(...);
```

deinhofer, 2016-11-29:
Here you can see, how it looks like before the enhancement: 
https://github.com/asterics/AsTeRICS-Ergo/blob/master/custom/bin/ARE/data/webservice/index.html#L90

mariokom, 2016-12-16:
Regarding the above, it is currently not possible to store the deployed model in ARE.

To retrieve the deployed model, AsapiSupport#getModel() function is used. This is the first line of the returned string (CameraMouse model):
<model xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" modelName="C:\AsTeRICS\bin\ARE\models\CameraMouse.acs_2015_10_9_14_13" **modelVersion**="20130320" **xsi:noNamespaceSchemaLocation="deployment_model.xsd"**>

This is the actual content when opening the 'CameraMouse.acs' file with a text editor:
<model xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" **xmlns:xsd="http://www.w3.org/2001/XMLSchema"** modelName="C:\AsTeRICS\bin\ARE\models\CameraMouse.acs_2015_10_9_14_13" **version**="20130320">

The two lines are different (other parts of the model are also different)
For some reason, the AsapiSupport#getModel() function returns an older version of the model and this is why we are unable to store the currently deployed model.

Any solution on this?

deinhofer, 2016-12-21:
Hmm, I actually don't understand why the method AsapiSupport#getModel() uses the file model.xml. Where is that file created?

On the other hand, does it have an impact on the functionality if the model representations are different?

deinhofer, 2017-01-23:
because there is a workaround: label it with low priority