asterics/AsTeRICS #143: REST: Add SSE subscription and notification for AsTeRICSErrorHandling messages

deinhofer, 2017-01-23:
When an Asterics plugin or the ARE itself has an error it should report it via [AstericsErrorHandling.reportError](https://github.com/asterics/AsTeRICS/blob/master/ARE/middleware/src/main/java/eu/asterics/mw/services/AstericsErrorHandling.java#L112)

The current REST API does not have the possibility to access the error information.

Specification:

1. Enhance SSE subscription to be able to subscribe for certain componentTypes|ARE|ALL and logging levels.

2. Broadcast reported messages to subscribed listeners.

deinhofer, 2017-10-20:
if there is time left. would generify error broadcasting to REST clients.

klues, 2017-10-20:
what do you my with "broadcasting to REST clients"? Whats the difference to just broadcasting to subscribed listeners?

deinhofer, 2017-10-20:
sorry for being unclear: I meant subscribed listeners