asterics/AsTeRICS #157: Test Asterics with Java 9 JRE

deinhofer, 2017-10-11:
JRE 9 is released. Test the ARE and APE for compatility to it.

benjaminaigner, 2017-10-19:
According to http://www.oracle.com/technetwork/java/javase/downloads/jdk9-downloads-3848520.html
no 32bit JVM is available for Java9, so no 32bit only plugin will work.

benjaminaigner, 2017-10-19:
Not working on Linux with Java9
version:  "Java HotSpot(TM) 64-Bit Server VM (build 9.0.1+11, mixed mode)"
-) No websockets/webserver

Typical plugins (camera mouse, Java Swing GUI,...) are working as expected on Linux.
Windows tests will follow.

deinhofer, 2017-10-19:
No websockets/webserver? Which version did you use for testing ?
Do you have an exception?

benjaminaigner, 2017-10-19:
No exception, current master, clean build.

deinhofer, 2017-12-14:
won't be fixed for release 3.0, as Java 8 works