asterics/AsTeRICS #163: Apply document root specification and move webserver document root to ARE/webservice

deinhofer, 2017-10-11:
As of https://github.com/asterics/AsTeRICS/wiki/AsTeRICS-webserver-document-root-specification

klues, 2017-10-13:
As I understood it we/you wanted to change the webserver document root to ARE/data?! Or did I understand it wrong? However in the link it states the webserver document root is `ARE/data/webserver/ ` and the title of this issue says `ARE/webservice`. I think this should be corrected/synchronized.

deinhofer, 2017-10-13:
sorry for this confusion:

currently the document root is 

ARE/data/webservice

The folder is below data because it originally was meant to hold configuration files for the webservice bundle and all such files are stored in the ARE/data folder.

But then I thought it would be nice to move the document root up to the ARE folder and then virtually map the `data` and the `models` folder to the document root.
Otherwise the document root would be mapped into the document root recursively.

We should also consider this for the storeData function

klues, 2017-10-16:
ok, so if ARE will be the document root of the webserver, what about the new folders (paths) defined in https://github.com/asterics/AsTeRICS/wiki/AsTeRICS-webserver-document-root-specification ?
e.g. the folders `webapps` or `modelSettings` - where will they be physically? will there be 2 new folders `ARE/modelSettings` and `ARE/webapps`?

Maybe we should add the information where the physical folder will be (if there is any) to the wiki-page.

klues, 2017-10-19:
My suggestion regarding saving configuration data of webapps:
I think the data should be physically located in `web/webapps/{name-of-webapp}/data` (other webapp data from package could be in `web/webapps/{name-of-webapp}/static`). So all data regarding a webapp is located in a single folder and not spread in different folders.

I think we then should change the REST-API to provide a command (path) like:
`/storage/webapp/{webappName}/{filepath}`
and the REST server checks if the given webapp exists and saves the given data to `web/webapps/{name-of-webapp}/data`.
The current method of https://github.com/asterics/AsTeRICS/pull/169 (line: https://github.com/klues/AsTeRICS/blob/d8475849b57262e0e439bea1706761fa19d2315d/ARE/services/WebService/src/main/java/eu/asterics/mw/webservice/RestServer.java#L768) then can be used to save data to the data-folder, e.g. with path:
`/storage/data/{filepath}`

@deinhofer: what do you think about this? should I add a special REST-method for saving webapp-data? And maybe also for getting the saved data (same path with GET instead of PUT)?

deinhofer, 2017-10-19:
The suggestion sounds good to me!

klues, 2017-10-19:
ok, then i will add such a new rest method to https://github.com/asterics/AsTeRICS/pull/169 before we merge it.

deinhofer, 2017-10-30:
this is optional for AsTeRICS Ergo but mandatory for Asterics

klues, 2017-10-30:
we will do this after https://github.com/asterics/AsTeRICS/issues/163 is done

PR #201 (2017-11-15): Implementation of changing web server document root (#163)
# This PR fixes #163.

* moved location of webserver document root from ARE/data/webservice/ to ARE/web according to the webserver document root specification (https://github.com/asterics/AsTeRICS/wiki/AsTeRICS-webserver-document-root-specification)
* changed build files accordingly

* added WebACS as submodule at ARE/web/webapps/WebACS/

## ResourceRegistry:
* added method getWebDocumentRootContentList which returns a list of URIs or resources contained in * the web document root folder
made constant for RECURSIVE_FILE_SEARCH_DEPTH

# [... cut]
commit bb9dadec (2017-11-15): Merge pull request #201 from asterics/new-docroot

Implementation of changing web server document root (#163)
Fixes #163
closed 2017-11-15 by deinhofer