asterics/AsTeRICS #185: Special character in bundle descriptor

ChrisVeigl, 2017-10-22:
Special characters in the bundle descriptor (e.g. german Umlaute in string constants for property default values) are not correctly transferred to the ACS via  "Download Component Collection".

The reason might be a wrong character encoding of the .xml file or during the processing/transfer.

deinhofer, 2017-10-23:
also consider XML special characters e.g. <> " /
They should automatically be quoted.

deinhofer, 2017-10-31:
all locations that open/store files on the ARE now use the respective methods from class ResourceRegistry:

**Method to get the contents of a file encoded in UTF8 or UTF8 with BOM**
https://github.com/asterics/AsTeRICS/blob/master/ARE/middleware/src/main/java/eu/asterics/mw/services/ResourceRegistry.java#L607

**method to store a resource encoded in UTF8**
https://github.com/asterics/AsTeRICS/blob/master/ARE/middleware/src/main/java/eu/asterics/mw/services/ResourceRegistry.java#L640

The implementations use the library apache-commons-io and the class BOMInputStream to correctly deal with UTF8 with and without BOM
http://www.rgagnon.com/javadetails/java-handle-utf8-file-with-bom.html
https://commons.apache.org/proper/commons-io/javadocs/api-2.5/org/apache/commons/io/input/BOMInputStream.html

@TODO: 
* The WebACS handles special characters correctly.

* The legacy C# ACS does **not correctly handle special characters** when they are downloaded from the ARE and are contained in a bundle descriptor.
* Additionally the C# ACS add the text 
```encoding=UTF-16```
to the xml string

deinhofer, 2017-10-31:
fixed in https://github.com/asterics/AsTeRICS/commit/c8f6ba3030d4ecbf4369b6cb5ae33289020c21bf and prior commits

closed 2017-10-31 by deinhofer