asterics/AsTeRICS #208: ARE installer - .dep dependencies + missing Debian packages

benjaminaigner, 2017-12-10:
is it really necessary to install the "libgstreamer0.10-dev" and "libgstreamer-plugins-base0.10-dev" packages, even on the already built ARE?

And on Debian, following packages not available:
'libgstreamer0.10-dev' 
'gstreamer0.10-tools'
'libgstreamer-plugins-base0.10-dev'
'gstreamer0.10-plugins-good'

Possible replacements maybe on gstreamer1.0?

deinhofer, 2017-12-11:
The dependencies were necessary on in order for javacv facetracking to work on Ubuntu 16.04. On Ubuntu 12.04 it worked without those packages. Maybe since I upgraded to javacv1.3 the legacy dependencies are not necessary any more.

I will test that.

deinhofer, 2017-12-14:
removed dependencies again.
seems to work, if not let's just add some documentation to the release notes.

closed 2017-12-14 by deinhofer