asterics/AsTeRICS #216: XFaceTracker does not Work with some cameras/OS

klues, 2017-12-15:
Tested AsTeRICS 3.0 on Windows and the XFaceTrackerMouse does not work in the following setups:
* Win10/Lenovo T410s/Integrated Camera
* Win7/PC/external Medion Camera

On my Lenovo T410s the camera now also is not working anymore with VLC, so maybe thats a general/other issue. But on the Win7 computer the Medion camera works with VLC, but not with XFaceTracker. However, the old camera mouse works.

klues, 2017-12-15:
attaching logs of the failed cases, interesting lines:
```
SETUP: Device is setup and ready to capture.

Dez 15, 2017 8:57:05 AM eu.asterics.mw.computervision.SharedFrameGrabber$GrabberThread run
WARNUNG: Could not start frame grabbing with grabber: org.bytedeco.javacv.VideoInputFrameGrabber@64b5a5, reason: videoInput.getPixels() Error: Could not get pixels.
```
[XCameraMouse-Win10-internalT410sCamera.txt](https://github.com/asterics/AsTeRICS/files/1562133/XCameraMouse-Win10-internalT410sCamera.txt)

deinhofer, 2017-12-15:
* On Lenovo devices there is a communications utility where you can enable/disable the camera generally. Try to disable/enable it.
* Also try what happens if you reboot your computer.
* Also try another FrameGrabber (OpenCV)
  * Edit model and set the framegrabber property to 'OpenCV'
  * Deploy and start model

klues, 2017-12-15:
Ok, I tried this on Win7 with the Medion camera:
* change FrameGrabber to "OpenCV"
* start Model -> did not work
* stop Model -> start Model -> did work!

so maybe "OpenCV" should be the standard on Windows?

deinhofer, 2017-12-15:
no, in almost all cases video input has better camera support at least that we experienced with the old plugin facetrackerLK. But the good thing is that now you have another option to try :-)

klues, 2017-12-15:
ok.
maybe it could be possible to automatically try other FrameGrabber if one does not work? Because it is not very intuitive for the user, if he has to edit to model to get the camera mouse work. 
(However it is now easier with F8 ;))