asterics/AsTeRICS #217: "sudo apt-get oracle-java8-installer" does not work

klues, 2017-12-15:
Tried to install AsTeRICS 3.0 on XUbuntu 16 and raspian PIXEL. In both cases the recommended command:
```
sudo apt-get oracle-java8-installer
```
to get the Java JRE did not work, the package was not found. Then used the openJDK package, which worked.
However, we should not recommend "oracle-java8-installer" if it does not exist.

deinhofer, 2017-12-15:
you must add a package source as described here: https://medium.com/coderscorner/installing-oracle-java-8-in-ubuntu-16-10-845507b13343

so maybe we should simply link to that tutorial?
The alternative is to directly use the download from oracle: http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html

klues, 2017-12-15:
ok, i thanks. i added this info to the release: https://github.com/asterics/AsTeRICS/releases/tag/v3.0

closed 2017-12-15 by klues