asterics/AsTeRICS #231: Jnativehook starting with error message on Win10

stefanstocki, 2018-03-20:
When starting are the following error message pops up:

[image]

ARE 3.0 on WIN10 Enterprise 64 bit 

problem since "Funktionsupdate Version 1709"

```
java.lang.NullPointerException
        at eu.asterics.mw.jnativehook.NativeHookServices.cleanupDLL(NativeHookServices.java:120)
        at eu.asterics.mw.jnativehook.NativeHookServices.<init>(NativeHookServices.java:80)
        at eu.asterics.mw.jnativehook.NativeHookServices.getInstance(NativeHookServices.java:180)
        at eu.asterics.mw.jnativehook.NativeHookServices.init(NativeHookServices.java:186)
        at eu.asterics.mw.jnativehook.NativeHookActivator.start(NativeHookActivator.java:36)
        at org.eclipse.osgi.framework.internal.core.BundleContextImpl$1.run(BundleContextImpl.java:783)
        at java.security.AccessController.doPrivileged(Native Method)
        at org.eclipse.osgi.framework.internal.core.BundleContextImpl.startActivator(BundleContextImpl.java:774)
        at org.eclipse.osgi.framework.internal.core.BundleContextImpl.start(BundleContextImpl.java:755)
        at org.eclipse.osgi.framework.internal.core.BundleHost.startWorker(BundleHost.java:370)
        at org.eclipse.osgi.framework.internal.core.AbstractBundle.start(AbstractBundle.java:284)
        at org.eclipse.osgi.framework.internal.core.AbstractBundle.start(AbstractBundle.java:276)
        at org.eclipse.osgi.framework.internal.core.FrameworkCommandProvider._start(FrameworkCommandProvider.java:252)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at org.eclipse.osgi.framework.internal.core.FrameworkCommandInterpreter.execute(FrameworkCommandInterpreter.java:155)
        at org.eclipse.osgi.framework.internal.core.FrameworkConsole.docommand(FrameworkConsole.java:156)
        at org.eclipse.osgi.framework.internal.core.FrameworkConsole.runConsole(FrameworkConsole.java:141)
        at org.eclipse.osgi.framework.internal.core.FrameworkConsole.run(FrameworkConsole.java:105)
        at java.lang.Thread.run(Unknown Source)
```

deinhofer, 2018-03-20:
According to the stack trace (NativeHookServices.java:120) the exception happens here: https://github.com/asterics/AsTeRICS/blob/master/ARE/services/JNativeHook/src/main/java/eu/asterics/mw/jnativehook/NativeHookServices.java#L120

tempDLLs can only be null if 
```java
File[] tempDLLs = tempDir.listFiles(new FilenameFilter()
```
returns null, which is the case if the given file is not a directory or an I/O error occured (see https://docs.oracle.com/javase/7/docs/api/java/io/File.html#listFiles(java.io.FilenameFilter)).

-->
Can other applications access your 
What is the value of your java.io.tempdir property? 
Is still space left on your drive?

klues, 2018-03-21:
On my computer there is no problem starting ARE 3.0 (or starting ARE from current master), although I also have installed Windows 10 Update "Funktionsupdate Version 1709".

deinhofer, 2018-03-27:
@stefanstocki Have you checked your tempdir and the space left? Do you still have the problem??

stefanstocki, 2018-03-27:
just tested again - now it starts without any problems...

closed 2018-03-27 by stefanstocki