asterics/AsTeRICS #245: ACS-Help does not display

ChrisVeigl, 2018-05-07:
When pressing F1 with a selected Plugin in the "legacy" ACS on Windows, the new Help System does not come up. The plugin-specific help pages should be displayed but the asterics help page appears in the browser with an error message that the help file cannot be found.

This might be related to Browsers which cannot display the help files from the local file system (when the ARE is not connected). 

It would be good to get the help files from the web in this case (as a fallback ?)

deinhofer, 2018-05-08:
Could you provide a screenshot?

If the reason is related to local file access restrictions of the browser then the easiest solution would be to use the [online help](http://asterics.github.io/AsTeRICS/help/index.html) as a primary source, if the ACS is not connected to the ARE and the help of the ARE if the ACS is connected to it.
The reason why we need this 2-level approach is that the online version only reflects the plugin-set of the current AsTeRICS version. In contrast, the help which is hosted on the ARE directly also contains pugin-help files of newly created plugins.

# Workaround
Use the [online help](http://asterics.github.io/AsTeRICS/help/index.html)

ChrisVeigl, 2018-05-08:
here a screenshot of the problem:

[image]

additional information: i just tried the help system on my lenovo win10 laptop and  is was unusable as well!  When pressing F1 in the legacy ACS (with a selected plugin) the Internet Explorer was opened (!) - and believe me - i did not use IE for years and it definitely is not my standard application for web pages ...   

also after connecting to the ARE, the help files could not be displayed.

after disconnection from the ARE, following error message was displayed when hitting F1:

[image]

(internet connection was present at any time)

When pressing F1 in the WebACS, nothing happens at all.

The help files are available on my systen in the local ARE subfolder /web/help 
the only way to use them is by manually opening them ...

From my perspective, the help system is currently broken.

deinhofer, 2018-05-15:
On my Win7 laptop with firefox 59.0.3 as standard browser everything works as expected.
* In legacy ACS: 
  * Press F1: The ACS-Help opens a local file within firefox
  * Connect to ARE, then press F1: The ACS-help opens the ACS-help from the ARE-webserver
  * Disconnect from ARE, then press F1: The ACS-help opens a local file within firefox
 
* In WebACS (http://asterics.github.io/AsTeRICS/webapps/WebACS/?areBaseURI=http://localhost:8081) opened in firefox:
  * Pressing F1: Opens a new tab with the ACS-help online versino (http://asterics.github.io/AsTeRICS/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)
  * Connect to ARE, Press F1: Opens a new tab with the ACS-help online versino (http://asterics.github.io/AsTeRICS/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)
  * Disconnect from ARE, Press F1: Opens a new tab with the ACS-help online versino (http://asterics.github.io/AsTeRICS/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)

* WebACS opened from ARE (http://localhost:8081/webapps/WebACS/index.html?autoConnect=true&autoDownloadModel=true)
  * Press F1: Opens ACS-help from ARE (http://localhost:8081/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)
  * Connect to ARE, Press F1: Opens ACS-help from ARE (http://localhost:8081/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)
  * Disconnect from ARE, Press F1: Opens ACS-help from ARE (http://localhost:8081/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)

So the following questions should be investigated:
* Why, in your case, the Internet Explorer (which is not supported) instead of the standard browser is used?

Obvisously the behaviour of legacy ACS and the WebACS is not the same, this should be unified to
1. If the ACS (WebACS) is not connected to the ARE: Open ACS-Help online (http://asterics.github.io/AsTeRICS/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)
2. If the ACS (WebACS) is connected to the ARE: Open the ACS-Help of the ARE to reflect the plugin set of the ARE (http://localhost:8081/webapps/WebACS/help/index.html?acs&ACS_Basic_Functions.htm)

klues, 2018-05-17:
On my computer the same behaviour: in lecacy ACS pressing `F1` opens the help in Internet Explorer.

deinhofer, 2018-05-28:
@ChrisVeigl 
@klues 
@sabicalija 
@stefanstocki 
@benjaminaigner 
@kingcole 

Please try the following to commands on the command line of your windows system and post
* browser used for opening
* windows version (incl. update)
* installed browsers (incl. version)
* and standard browser

1. Try explorer cmd
```cmd
explorer http://asterics.github.io/AsTeRICS/help/index.html
```
2. Try start cmd
```cmd
start http://asterics.github.io/AsTeRICS/help/index.html
```

klues, 2018-05-28:
1. opens in Firefox (my standard browser)
2. opens in Firefox (my standard browser)

Windows 10 Education, Build 16299.431
Firefox 60.0.1

klues, 2018-05-28:
I  think we should just return `explorer` or `start` in this method:
https://github.com/asterics/AsTeRICS/blob/master/ACS/ACS/MainWindow.xaml.cs#L7931
after we finished this test.

deinhofer, 2018-05-28:
Win 7 Enterprise, SP1, 64bit

# installed browsers
 * firefox 60.0.1 (**standard**)
 * chrome 66.0.3359.181
 * Internet explorer 11 

1. opens firefox 60.0.1
2. opens firefox 60.0.1

ChrisVeigl, 2018-05-28:
System: 
  Lenovo Thinkpad 51s
  Windows 10 Pro 64-bit (2018) 
  
Browsers: 
  Firefox 60.0.1 (standard)
  Chrome Version 66.0.3359.181
  Microsoft Edge 42.17134.1.0
  Internet Explorer 11

1: help system opens correctly in Firefox
2: help system opens correctly in FireFox

(will try the other system  - where the problem occured - later)

kingcole, 2018-05-28:
Windows 10 Enterprise, 64bit, Version 1803, build 17134.48
Firefox 60.0.1 (Standard)
Google Chrome Version 66.0.3359.181 (Offizieller Build) (64-Bit)
Safari 5.1.7 (7534.57.2)
Opera 52.0.2871.64
Microsoft Edge 42.17134.1.0

opened help correctly in Firefox in both cases

ChrisVeigl, 2018-05-28:
System2:
Lenovo Yoga Pro 
Windows 8.1 64-bit (2013)

Browsers:
Firefox 60.0.1 
Chrome Version 66.0.3359.181 (standard)
Internet Explorer 11

1: help system opens correctly in Chome
2: help system opens correctly in Chrome

This is the System which causes problems in the ACS 
- when pressing F1, the ACS opens the Internet Exporer which cannot display the help content...

klues, 2018-05-29:
it seems that both `start` and `explorer` opens the standard browser in both cases. So I think the issue should be resolved if we just return one of them in this method instead of the registry-browser-search:
https://github.com/asterics/AsTeRICS/blob/master/ACS/ACS/MainWindow.xaml.cs#L7931

deinhofer, 2018-05-29:
Let's fix the opening of the standard browser as @klues  suggested. Instead of searching in the registry use explorer or start cmd.
In the ARE and the ApplicationLauncher plugin, the actual command used to open a website is configured in areProperties, e.g.
```
# Default Windows command to start a browser.
ARE.openURL.cmd.windows=C:/Users/mad/AppData/Local/Google/Chrome/Application/chrome.exe
# Default Mac OSX command to start a browser.
ARE.openURL.cmd.macosx=open
# Default Linux command to start a browser.
ARE.openURL.cmd.linux=sensible-browser
```
which works pretty good. Should we do the same here? This would allow the user to configure a browser of choice as well?

This issue was split up into two. Let's use this one to fix the browser open problem of the legacy ACS. #252 describes how to fix the strategy of which help files are actually shown.

ChrisVeigl, 2018-05-29:
Sounds like a good solution!

If it is possible to detect the problem during page display, it would be nice to show a message how to configure/change the Webbrowser

klues, 2018-05-30:
should be solved with
https://github.com/asterics/AsTeRICS/pull/253/files
please try out:
```
git pull
git checkout bklaus/issue#252/unify-strategy-to-show-help
//start file AsTeRICS\bin\ACS\ACS.exe
```
`F1` should open online help, and if connected to ARE the help from ARE.

klues, 2018-05-30:
in `asterics.ini` there are now two new parameters:
```
acsOnlineHelpBaseURL=http://asterics.github.io/AsTeRICS/
helpBrowserCommand=explorer
```
`acsOnlineHelpBaseURL` is the path to the online help.
`helpBrowserCommand` is the command to open the browser, could be replaced by absolute path to another browser (e.g. Chrome) if required.

deinhofer, 2018-06-07:
Fixed with #253

closed 2018-06-07 by deinhofer