asterics/AsTeRICS #246: Model file collection not available under Linux/Mac

ChrisVeigl, 2018-05-07:
The majority of model files, use case demos and component test models is stored in the folder  ACS/models. These models are also referenced from the user manual (e.g. in the turorials section).

The whole ACS folder is not present in the Linux and Mac distributions - so all these models are missing.

It would be reasonable to copy the models subfolders from ACS/models to ARE/models 
(in the Linux and Mac distributions)

deinhofer, 2018-05-08:
This should be very easy to achieve, but I would suggest to directly move the models in the repository and do the change for all distributions.
I don't like exceptions, the produce work load ;-)

For a long term goal, we should provide the model files online on the info and demo page and link from the plugin help files to respective example model files.

ChrisVeigl, 2018-05-08:
okay - we could move all model files to the ARE folder and change the references in the user manual!
maybe also a good time to remove obsolete models ...

klues, 2018-05-09:
I also would suggest to simply move all models to the `ARE` folder. It's less confusing.