asterics/AsTeRICS #269: Clicking on edit model (F8) does not open WebACS with current model on Win10

deinhofer, 2018-11-16:
In the current master branch the file explorer is opened instead. In Asterics 3.0.1 it still works correctly.

Here is the log output for the master branch
```
2018-11-16 20:01:34.830 FINE [OSUtils startApplication] adding argument: http://localhost:8081/webapps/WebACS/index.html?autoConnect=true&autoDownloadModel=true
2018-11-16 20:01:34.830 FINE [OSUtils startApplication] Finally constructed command: [explorer, http://localhost:8081/webapps/WebACS/index.html?autoConnect=true&autoDownloadModel=true]
```

The log output for 3.0.1:
```
Nov 16, 2018 8:03:05 PM eu.asterics.mw.utils.OSUtils startApplication
FEIN: adding argument: "http://localhost:8081/webapps/WebACS/index.html?autoConnect=true&autoDownloadModel=true"
Nov 16, 2018 8:03:05 PM eu.asterics.mw.utils.OSUtils startApplication
FEIN: Finally constructed command: [explorer, "http://localhost:8081/webapps/WebACS/index.html?autoConnect=true&autoDownloadModel=true"]
```
It seems that the missing quote of the command parameter (the URL) is responsible for the problem.
This was probably broken by https://github.com/asterics/AsTeRICS/commit/0b0cf23cf163184de35756a6393d0bd3c7734097

# Solution
Simply surrounding the arguments with quotes is probably no good idea, as the OSUtils class cannot know the meaning of the arguments (if there should be 1 or several seperated by space). 
Should we fix the invocation of of OSUtils in ApplicationLauncher Plugin? Are there any side effects on other platforms?

@klues  any ideas?

klues, 2018-11-21:
see #270 for my solution. please review and merge if you agree.

klues, 2018-11-26:
Testcases model files for Application launcher: 
[ApplicationLauncher-Tests.zip](https://github.com/asterics/AsTeRICS/files/2615891/ApplicationLauncher-Tests.zip)

deinhofer, 2018-11-27:
Fixed by #270

closed 2018-11-27 by deinhofer