asterics/AsTeRICS #282: Deadzone events "exitZone" and "enterZone" wrong documented/implemented?!

klues, 2019-01-24:
I am confused about the events `exitZone` and `enterZone` of the Deadzone plugin:

Documentation says:
http://asterics.github.io/AsTeRICS/help/index.html?plugins&processors//Deadzone.htm

> 
    enterZone: This event is triggered when the x or x- and y- values enter the specified radius around the centre.
    exitZone: This event is triggered when the x or x- and y- values leave the specified radius around the centre.

Implementation is this:
https://github.com/asterics/AsTeRICS/blob/56f59f844f2a1c038ff4807ee208127d2033298b/ARE/components/processor.deadzone/src/main/java/eu/asterics/component/processor/deadzone/DeadzoneInstance.java#L264
`etpEnterZone.raiseEvent();` ist called if `dist >= propRadius`, so if the values **leave** the radius around the centre. This is the opposite of the documentation, isn't it? The same applies to the `exitZone` event.

klues, 2019-01-24:
@ChrisVeigl @deinhofer isn't this a bug?

deinhofer, 2019-01-24:
yes, I think you are right. Looks like a copy&paste failure.
But how to fix it backwards compatible? Older models might use it this way.
We could correct also in the models which are in the Asterics repository, but it would still be a problem, if you want to deploy a fixed model to an older ARE instance e.g. 3.0.

PR #286 (2019-02-10): fixed documentation of Deadzone plugin
fixed documentation, corrected definition of 'enterZone' and 'exitZone' event as well es definition of 'deadzone' mode.

fixes #282
closed 2019-02-10 by deinhofer