asterics/AsTeRICS #318: CrossHair Plugin won't start: "TRANSLUCENT translucency is not supported"

benjaminaigner, 2019-05-09:
If I try to start a model with the crosshair plugin, it won't start.
Error message:
[image]

System:
Debian Stretch (stable), x86-64
Cinnamon Desktop (lightdm)
AsTeRICS 4.0.0-alpha1

klues, 2019-05-10:
Thanks for filing the bug. Could you please also post the stacktrace so that I can see where the exception is thrown?

benjaminaigner, 2020-06-02:
Sry for missing your request.

AsTeRICS Version: 4.1.0
Here are both stacktraces, for the Crosshair and the Tooltip Plugin:

```
2020-06-02 09:40:32.750 SEVERE [DeploymentManager runModel] java.lang.UnsupportedOperationException: TRANSLUCENT translucency is not supported.
        at java.awt.Window.setOpacity(Window.java:3627)
        at java.awt.Frame.setOpacity(Frame.java:962)
        at eu.asterics.component.actuator.tooltip.GUI.<init>(GUI.java:73)
        at eu.asterics.component.actuator.tooltip.TooltipInstance.start(TooltipInstance.java:321)
        at eu.asterics.mw.are.DeploymentManager.runModel(DeploymentManager.java:582)
        at eu.asterics.mw.services.AREServices.runModelInternal(AREServices.java:440)
        at eu.asterics.mw.services.AREServices$7.call(AREServices.java:409)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```

```
2020-06-02 09:40:32.740 SEVERE [DeploymentManager runModel] java.lang.UnsupportedOperationException: TRANSLUCENT translucency is not supported.
        at java.awt.Window.setOpacity(Window.java:3627)
        at eu.asterics.component.actuator.crosshaircursorcontrol.GUI.<init>(GUI.java:90)
        at eu.asterics.component.actuator.crosshaircursorcontrol.CrosshairCursorControlInstance.start(CrosshairCursorControlInstance.java:562)
        at eu.asterics.mw.are.DeploymentManager.runModel(DeploymentManager.java:582)
        at eu.asterics.mw.services.AREServices.runModelInternal(AREServices.java:440)
        at eu.asterics.mw.services.AREServices$7.call(AREServices.java:409)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)

```

klues, 2020-06-02:
branch `klaus/issue#318/translucency-not-supported` should fix it, please verify.
Afterwards we can merge #344

PR #344 (2020-06-02): prevent crash if setOpacity not supported
fixes #318
closed 2020-06-02 by klues