asterics/AsTeRICS #320: some help files can't be accessed anymore

ChrisVeigl, 2019-06-07:
I just found that the pressing "F1" in the modelSwitcher plugin results in a 404 error, although the page https://www.asterics.eu/plugins/actuators/ModelSwitcher.html exists.

The problem is that the ACS tries to access the page https://www.asterics.eu/plugins/actuators/modelSwitcher.html
(small "m") and the filesystem is case-sensitive.

The same problem exists for plugins that do not follow the naming convention and start with a non-captial letter: serialSender, enocean, openHAB, openBCI, p2_parser.

I suggest to rename the help files for those plugins (or to refactor the plugin name in bundle descriptors and source files - but it should be verified that this does not cause problems with existing models)

sabicalija, 2019-06-07:
Great suggestions! Thanks a lot.

Alternatively, we could save the link, to the documentation, in the component collection in a dedicated entry. I think this way, the procedure is more robust (less error-prone) and not that susceptible to changes.

deinhofer, 2021-03-16:
The fix works for most of the incosistent plugin names mentioned above, but there are two who still don't work:

* p2_parser looks for P2_parser.html but the help file is P2_Parser.html
* IRMicro looks for IRMicro.html but the help file is IrMicro.html
* IrTrans looks for IrTrans.html but the help file is IRTrans.html

Maybe there are still others. So I think in this case the simplest solution would really be to rename the help filename in the docs folder

klues, 2021-03-17:
> So I think in this case the simplest solution would really be to rename the help filename in the docs folder

Yeah, why not 👍 
Maybe this is even easier: https://stackoverflow.com/a/14814577/9219743

deinhofer, 2021-04-26:
> > So I think in this case the simplest solution would really be to rename the help filename in the docs folder
> 
> Yeah, why not +1
> Maybe this is even easier: https://stackoverflow.com/a/14814577/9219743

This would be the easiest solution indeed, but unfortunately, we don't have access to the webserver. In order for that we would have to move the hosting to another server, e.g. ionos.

commit 0d29b0fb (2020-04-24): * Fixes #320 and loads component collection of ARE by default

    Ensures that the first characater is always uppercase when creating the URL help string.
    This should work for all mentioned plugins (serialSender, enocean,...)
    Only P2_Parser.html should be renamed to P2_parser.html

* set startup bundle to ..\ARE\web\componentCollections\defaultComponentCollection.abd to load directly the component collection which was generated by the ARE at startup. This should ensure that the ACS is in sync with the local ARE by default.
commit dc658ad7 (2020-04-24): Fixes #320

Ensures that the first characater is always uppercase when creating the URL help string.
This should work for all mentioned plugins (serialSender, enocean,...)
Only P2_Parser.html should be renamed to P2_parser.html
commit dfeb2323 (2020-05-04): Commited Fixes #320 again, did not worn on other machine