asterics/AsTeRICS #336: Notes on installing .deb packets

benjaminaigner, 2020-05-04:
If the asterics-are.deb is installed via:
`dpkg -i asterics-are-4.1.0-alpha3.deb`
you get following error (if the necessary packages are not installed):
```
dpkg: dependency problems prevent configuration of asterics-are:
 asterics-are depends on libhidapi-libusb0; however:
  Package libhidapi-libusb0 is not installed.
```
It is possible to install the libraries afterwards with:
`sudo apt --fix-broken install`

I think it will be good to have the information on the installation page.

In addition, I would replace all the information on the release page by a link to
https://www.asterics.eu/get-started/Installation.html

deinhofer, 2020-05-04:
Thanks, good idea. Will be in next publishing.

ChrisVeigl, 2020-05-04:
could we add `libhidapi-libusb0` in the installer to prevent the dependency problems?

deinhofer, 2020-05-04:
You cannot directly add the binary package, but the package name, which is the case, see https://github.com/asterics/AsTeRICS/blob/master/Installer/asterics-are/package/linux/control.

I have updated the installation docs, will look like this: https://www.asterics.eu/next/get-started/Installation.html#linux

If you want to change it, click onto the ```Edit this page``` link on the bottom of the page.

ChrisVeigl, 2020-05-04:
could we do something like
`sudo apt install libhidapi-libusb0 &&  dpkg -i asterics-are-<version>.deb` ?

benjaminaigner, 2020-05-05:
@ChrisVeigl This would be the better way, no error messages there :-).
Small correction (sudo for dpkg, wildcard for easy copy&paste):
`sudo apt install libhidapi-libusb0 && sudo dpkg -i asterics-are-*.deb`

THX @deinhofer

closed 2020-05-05 by benjaminaigner