asterics/AsTeRICS #337: [javac] warning: bootstrap class path not set in conjunction with -source 1.7

kalimero0073, 2020-05-05:
Dear community,

I receive following compiler problem when running one newly created plugin's build.xml:

> [javac] warning: bootstrap class path not set in conjunction with -source 1.7

> [javac] (use -source 8 or higher to enable lambda expressions)

I configured my computer and eclipse to use jdk 1.8, but although I can use lambda expressions in the IDE it does not compile with ant.

kind regards,
Kale

klues, 2020-05-05:
Hi, if you open the commandline (e.g. "cmd") and type `java -version` what is the result? It should something like `java version "1.8...`.

kalimero0073, 2020-05-05:
C:\>java -version
java version "1.8.0_192"
Java(TM) SE Runtime Environment (build 1.8.0_192-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.192-b12, mixed mode)

klues, 2020-05-05:
Ok, thanks, that's strange. What if you try to start `ant run` in the base directory of AsTeRICS Grid within the commandline, not the IDE? Same error?

sabicalija, 2020-05-05:
What about `JAVA_HOME`? Does this need to be setup correctly as well?

klues, 2020-05-05:
These are my environemnt variables:
1) JAVA_HOME: http://prntscr.com/sb8h3b
2) PATH for ant: http://prntscr.com/sb8ht5
3) PATH for java: http://prntscr.com/sb8i8t

kalimero0073, 2020-05-05:
JAVA_HOME C:\Tools\Java\jdk1.8.0_192_32bit
ANT_HOME C:\Tools\apache-ant-1.10.5-bin\apache-ant-1.10.5
Path for ant: C:\Tools\apache-ant-1.10.5-bin\apache-ant-1.10.5\bin
PATH for java: C:\Tools\Java\jdk1.8.0_192_32bit\bin

the error links me to the plugin's build.xml
`  <target name="compile" depends="init" description="compile the source ">
    <javac source="${javac.source}" target="${javac.target}" includeantruntime="false" srcdir="${src.java}" destdir="${build}" verbose="true" debug="${debug}" 
           classpath="${classpath}"> <classpath refid="asterics.classpath"/>
     </javac>
  </target>`

So I thought there may be some default jdk value set somewhere, but I cannot localize it.

klues, 2020-05-05:
Hmm, the only difference I see is PATH for java on my system is `.../jre/bin` but for compiling this shouldn't make a difference?!

Otherwise unfortunately don't have more ideas, maybe @deinhofer ?

`warning: bootstrap class path not set in conjunction with -source 1.7` is only a warning, so maybe things nevertheless work?

kalimero0073, 2020-05-05:
I hardcoded it into the build file and it works now
`  <target name="compile" depends="init" description="compile the source ">
    <javac source=**"1.8"** target=**"1.8"** includeantruntime="false" srcdir="${src.java}" destdir="${build}" verbose="true" debug="${debug}" 
           classpath="${classpath}"> <classpath refid="asterics.classpath"/>
     </javac>
  </target>`

Thank you for you help! I very much appreciate it.... Kind regards, Kale

closed 2020-05-05 by kalimero0073

klues, 2020-05-05:
OK, thanks for posting your solution!