asterics/AsTeRICS #340: Getting the right jars - incl transitive dependencies

kalimero0073, 2020-05-07:
Dear community,

As I am very new to working with asterics and OSGI I am figuring out a way to include the right jars for developing a specific asterics plugin (I'd like to make a plugin that would use the javalin web framework - javalin.io)

When using maven, the dependencies are well managed. However, I downloaded all dependencies (incl transitives) with 
`mvn dependency:copy-dependencies -Dclassifier=sources`

I adapted the manifest and build file accordingly, built successfully the plugin, but I get a 

> java.lang.NoClassDefFoundError: kotlin/NoWhenBranchMatchedException

So I assume that I may follow a wrong process for getting the right jars.

From reading the developer manual I could not find an answer to this.
I hope I did not miss some apparent solution or guideline for this, if so I am sorry for bothering you.

Kind regards, Kale

klues, 2020-05-11:
I'm afraid I cannot really help for this. All I know is that:
1) Unfortunately AsTeRICS doesn't use a good dependency management system like maven. All dependencies are checked into this repository which is also the reason that it's so big. I don't know how exactly this came about, it was before I was envolved in this project.
2) Dependencies are directly checked in into the packages of the components. E.g. [here](https://github.com/asterics/AsTeRICS/tree/master/ARE/services/JNativeHook/src/main/resources/lib/native) you can see a `jar` that is used e.g. in the `KeyCapture` plugin, imported by `Require-Bundle: eu.asterics.mw.jnativehook` in `MANIFEST.MF`, see [here](https://github.com/asterics/AsTeRICS/blob/master/ARE/components/sensor.keycapture/src/main/resources/META-INF/MANIFEST.MF).

@deinhofer should know more about all this, maybe he can help.
If you find a way to integrate maven dependencies in a good way with the existing workflow, we would be definitely interested how you do it.

Just out of in interest: How do you know about AsTeRICS and in which context you want to use it?

kalimero0073, 2020-05-11:
Thank you very much for your reply! 

About me ...
I started working at the KI-I (https://ki-i.at/), with who I assume you are regularly in contact with.
For a project (GUIDed) I am exploring asterics, its plugins and the development of further plugins for some use cases.

klues, 2020-05-11:
Ah, ok, so you're not from an unknown organization. We already thought someone somewhere far away has found AsTeRICS by accident 😄

deinhofer, 2020-05-11:
Hi,
@kalimero0073 good to hear that there is some progress :-)
You are lucky, I have recently added the libs of [maven-resolver-ant-task](https://maven.apache.org/resolver-ant-tasks/) to easily dowload the dependencies of javacv and [bnd-tools](https://bndtools.org/) to easily create osgi-fied jars:

```ant
	<!-- taskdef bnd, unfortunately it's very bad documented -->
	<taskdef resource="aQute/bnd/ant/taskdef.properties" classpath="${biz.aQute:bnd:jar}"/>

	<!-- taskdef maven-resolver-ant-tasks -->
	<taskdef uri="antlib:org.apache.maven.resolver.ant" resource="org/apache/maven/resolver/ant/antlib.xml">
		<classpath>
		  <fileset dir="${dev.libs}" includes="*.jar"/>
		</classpath>
	</taskdef>

	<!-- define repositories for maven -->
	<resolver:localrepo dir="${javacv.dest}"/>
	<resolver:remoterepo id="central-https" url="https://repo1.maven.org/maven2/" type="default"
	          releases="true" snapshots="false" updates="daily" checksums="fail"/>
	<resolver:remoterepos id="resolver.repositories">
		<resolver:remoterepo refid="central-https"/>
	</resolver:remoterepos>
	
	<!-- use javacv pom file for dependency resolving -->
	<resolver:pom file="pom.xml" id="pom"/>
	
	<target name="download-javacv" description="Downloads the javacv jars">		
		<resolver:resolve>
			<dependencies pomRef="pom">
				<!-- exclude all unneeded artifacts here -->
				<exclusion coords="org.bytedeco:flycapture"/>
				<exclusion coords="org.bytedeco:flycapture-platform"/>
				<exclusion coords="org.bytedeco:ffmpeg"/>
				<exclusion coords="org.bytedeco:ffmpeg-platform"/>
				<exclusion coords="org.bytedeco:artoolkitplus"/>
				<exclusion coords="org.bytedeco:artoolkitplus-platform"/>
				<exclusion coords="org.bytedeco:flandmark"/>
				<exclusion coords="org.bytedeco:flandmark-platform"/>
				<exclusion coords="org.bytedeco:leptonica"/>
				<exclusion coords="org.bytedeco:leptonica-platform"/>
				<exclusion coords="org.bytedeco:libdc1394"/>
				<exclusion coords="org.bytedeco:libdc1394-platform"/>
				<exclusion coords="org.bytedeco:tesseract"/>
				<exclusion coords="org.bytedeco:tesseract-platform"/>
				<exclusion coords="org.bytedeco:librealsense"/>
				<exclusion coords="org.bytedeco:librealsense-platform"/>		
				<exclusion coords="org.bytedeco:librealsense2"/>
				<exclusion coords="org.bytedeco:librealsense2-platform"/>		
				<exclusion coords="org.bytedeco:libfreenect"/>
				<exclusion coords="org.bytedeco:libfreenect-platform"/>		
				<exclusion coords="org.bytedeco:libfreenect2"/>
				<exclusion coords="org.bytedeco:libfreenect2-platform"/>				
			</dependencies>
			<!-- downloads the resolved artifacts and stores them at ${javacv.dest} -->
			<files refid="api.files" dir="${javacv.dest}"
			       layout="javacv-platform.jar"/>
		</resolver:resolve>
	</target>	
``` 

So, you could do it the same way for your plugins. 
If this works, could you please update the [developer documentation](https://www.asterics.eu/develop/plugin/)?
[Here](https://www.asterics.eu/get-involved/Contribute.html) is how to do it.

kalimero0073, 2020-05-11:
@klues Yeah in this regard there is nothing special about me :D 

@deinhofer Hi! Thank you very much for the description! I will study, test it and include it in the documentation as soon as it works!

deinhofer, 2021-03-16:
answered

closed 2021-03-16 by deinhofer