asterics/AsTeRICS #342: AsTeRICS homepage: model description of Switch Mouse Solution

ChrisVeigl, 2020-05-11:
The model description of the Switch Mouse Solution contains a wrong information in the "requirements" section. here "cursor left" and "cursor right" should be replaced by "cursor right" and "cursor down".

maybe it would be more intuitive for provide a demo model which uses 4 keys (up, down, left, right) and add information about 2 or one switch models.

ChrisVeigl, 2020-05-11:
fixed with 26bbf89cc78dd7c94236d1c8b9b73264263f1bce

closed 2020-05-11 by ChrisVeigl

deinhofer, 2020-05-12:
Thanks for your update. I added model links that open in the webacs webapp for 1-switch and 4-switch models.
See how to do it: https://github.com/asterics/asterics-docs.js/blob/master/packages/@asterics-docs/vuepress-theme-docs/README.md#model-links