asterics/AsTeRICS #346: ACS crash when opening model

ChrisVeigl, 2021-04-18:
just found a very strange bug in the ACS: when opening certain models, e.g.:
https://github.com/asterics/AsTeRICS/blob/master/bin/ARE/models/useCaseDemos/mouseControl/crosshairCursorControl_1key.acs

the ACS crashes with attached error message:

[image]

This does not happen every time, and not on every computer (i have it repeatably on my Desktop PC with a large screen - maybe related to the problem)...

This does not happen in the WebACS.

klues, 2021-04-19:
Strange, never happened to me.
I think this is the code of interest:
https://github.com/asterics/AsTeRICS/blob/master/ACS/ACS/MainWindow.xaml.cs#L6041

`dockEventGrid.Width = dockableComponentProperties.ContainerPane.ActualWidth - 15;`
If `ContainerPane.ActualWidth` is `0`, the result width is `-15`, which is stated in the error message. No idea what the exact meaning is or why `ContainerPane.ActualWidth` is `0` sometimes.

ChrisVeigl, 2021-04-19:
thanks for having a look!
could we catch that exception - so that we prevent the ACS from crashing completely ... ?
I have no idea how often (or in which system constellation) the problem occurs - but if it happens, the model can't be used with the ACS at all ...

klues, 2021-04-20:
Yeah, would make sense to catch the exception or to prevent it by using a default value if `0` is returned. However I've no clue what this code is actually doing, this should be investigated before changing something here.