asterics/AsTeRICS #354: ACS: double-click opening .acs file -> cannot deploy model

benjaminaigner, 2021-06-25:
It seems, that the ACS on Windows 10 is opened with the wrong execution path,
when associated with .acs files.

Steps to reproduce:
* Select an .acs file, click _open with_
* Because the ACS is not listed in this menu, select _more apps_ and search for the ACS binary
* When double clicking the .acs file, the ACS is opened as intended
* Everything works, connecting to ARE and editing

Problem:
* Clicking "Upload model" is responded with "Error deploying model"
* In the .acs file's folder, a new text file is created, named _ACS.txt_
* Error:``` Die Datei "C:\Users\FH\<path to folder>\deployment_model.xsd" konnte nicht gefunden werden.```

It looks like the ACS is trying to open the xsd file in the .acs file folder instead of the installation folder.

deinhofer, 2021-06-25:
Hi,
yes I can confirm the bug and the cause. There are already two workarounds for it. 

## Workaround 1

If you assign the file extension .acs with the ACS.bat program, the problem ist solved.
The batch-file changes the execution directory.
[image]

## Workaround 2

Create a shortcut of ACS.exe and check the execution path in "Properties". The assign the .acs file extension to the shortcut.
[image]

benjaminaigner, 2021-06-25:
THX for the workarounds!