asterics/AsTeRICS #367: Account rotella gira

GiovannaDalBen, 2024-11-23:
Dopo aggiornamento, gli account non funzionano, la rotella gira.
Non riesco più a entrare nel sito.
Cosa fare?

klues, 2024-11-26:
Hello,
we had some server issues on the weekend, but this should be resolved now. Is it still not working? Does login on another device work?

GiovannaDalBen, 2024-11-26:
Sisi ora funziona. Grazie mille!!
Giovanna

Dott.ssa Giovanna Dal Ben
*Logopedista *
*Servizio Età Evolutiva*
*Via Torni, 51 - Mogliano Veneto (TV)*
*Distretto di Treviso Sud*
*Tel. 041-5988232*

-- 
*CLAUSOLA DI RISERVATEZZA:* Il contenuto della presente comunicazione è 
strettamente riservato, essendo indirizzato esclusivamente al destinatario 
sopra individuato e potendo contenere informazioni strettamente personali 
e/o confidenziali. Qualora fosse pervenuto a soggetto diverso dal 
destinatario questi deve intendersi sin d'ora avvisato che qualsiasi forma 
di diffusione dei dati, dei fatti e delle notizie apprese è assolutamente 
vietata. Si chiede cortesemente di cancellare il messaggio erroneamente 
ricevuto dal proprio sistema, dopo aver notificato al mittente l'errore 
commesso.

klues, 2024-11-26:
Great!

closed 2024-11-26 by klues